DLR-SC/tigl #1012: Python wrapper returns SwigPyObject instead of CPACSWalls

MarAlder, 2024-06-17:
I still can't figure out why the walls (CPACSWalls.h) are not being exported correctly with the Python bindings. Does anyone have an idea?

Minimal test code:
```Python
from tixi3.tixi3wrapper import Tixi3
from tigl3.tigl3wrapper import Tigl3
import tigl3.configuration

tixi_h = Tixi3()
tigl_h = Tigl3()

tixi_h.open("test_fuselage_walls_ex1.xml")
tigl_h.open(tixi_h, "")

mgr = tigl3.configuration.CCPACSConfigurationManager_get_instance()
aircraft_config = mgr.get_configuration(tigl_h._handle.value)
walls = aircraft_config.get_fuselage("fuselage").get_structure().get_walls()

help(walls)
```
Test data:
[test_fuselage_walls_ex1.zip](https://github.com/user-attachments/files/15863178/test_fuselage_walls_ex1.zip)

Output:
```
class SwigPyObject(object)
 |  Swig object carries a C/C++ instance pointer
 |
 |  Methods defined here:
 |
 |  __eq__(self, value, /)
 |      Return self==value.
 |
 |  __ge__(self, value, /)
 |      Return self>=value.
 |
 |  __getattribute__(self, name, /)
 |      Return getattr(self, name).
[...]
```

https://github.com/DLR-SC/tigl/blob/f1e1bc31633a0bed05f3d6c45963d5048320e5df/bindings/python_internal/configuration.i#L190-L191

joergbrech, 2024-06-17:
Is there an `#include "generated/CPACSWalls.h"` missing in `configuration.i`? Note the `#` instead of the `%`.

MarAlder, 2024-06-17:
I also thought so, but no effect (see 2e43950)

joergbrech, 2024-06-17:
Sorry, could you try `CCPACSWalls` instead of `generated/CPACSWalls.h`?

MarAlder, 2024-11-28:
Nice to see progress here, @merakulix! Perhaps we should strive to also cover such Python functions via unittests? I made an [example](https://github.com/MarAlder/tigl/blob/tanks/tests/unittests/test_fuelTanks.py) on my fork for the tanks. The pythonwrapper tests are still at the old TiGL/TiXI versions.

MarAlder, 2024-11-28:
True, let's move this discussion to #826.

PR #1035 (2024-11-28): Fix Python wrapper returns SwigPyObject instead of CPACSWalls (Issue #1012) 

CCPACSWalls couldn't be processed correctly via swig, due to missing include statements in configuration.i.
Adding those includes fixes the issue.

## Description

All required includes in configuration.i were added, which are:

- The header files of the C++ classes that need to be included
- all header files of classes, that deal with types of "boost-optional" or objects related to "ITiglGeometricComponent" that swig can't process otherwise

The changes fixes #1012 and fixes #986.

## How Has This Been Tested?

Runn [... cut]
closed 2024-11-28 by joergbrech