DLR-SC/tigl #1032: Wing cutout geometry with leading edge devices

hardt-fa-flb, 2024-11-08:
Hello everyone,

I have a question regarding the leading edge devices: is it possible to generate a wing geometry from TIGL in which the cutouts for trailing  AND leading edge devices are taken into account? The  `CCPACSWing::GetLoftWithCutouts` method unfortunately only considers the trailing edge devices. Here is my use case:

[tigl-slats.zip](https://github.com/user-attachments/files/17680683/tigl-slats.zip)

Best regards,

Daniel

joergbrech, 2024-11-08:
The leading edge devices are unfortunately not implemented yet.

daha96, 2024-11-11:
I'm currently working on a project which needs the cut outs for LE and TE devices, but that project is not DLR related. We would need the functionality till the end of the year. I assume with this background I need to implement a workaround for this issue?

joergbrech, 2024-11-11:
> [ ... ]  We would need the functionality till the end of the year. I assume with this background I need to implement a workaround for this issue?

Yes, I am afraid so. I am happy to help in figuring something out. It all depends on how complex the cutouts have to be. In the simplest case, you could just have a Boolean Cut with a box, as in https://github.com/rainman110/tigl-workshop/blob/master/Exercises/solutions/Exercise2_Customization_Visualization_Solution.ipynb.

daha96, 2024-11-11:
Thanks for the workshop notebook! I think I'll use it for the workaround

hardt-fa-flb, 2024-11-11:
Thanks for the workshop notebook! I think I'll use it for the workaround.