DLR-SC/tigl #1037: Fuslage CPACSSkin python binding seems to be broken

hardt-fa-flb, 2024-11-27:
I tried to access the fuselage skin segments with python. Unfortunately, I got stuck by walking through the class hierarchy at the `CPACSFuselageStructure.get_skin` method, that returns an empty object. The output of the script below is:

 `['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__int__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'acquire', 'append', 'disown', 'next', 'own']`.

Use case here: [fuselage.zip](https://github.com/user-attachments/files/17936616/fuselage.zip)

```python
from __future__ import print_function

from tixi3 import tixi3wrapper
from tigl3 import tigl3wrapper
import tigl3.configuration
import os

def display_fuselage_structure(tigl_handle):
    """
    This is an example how to use the internal tigl/pyocc API
    to display all wing and fuselage segments
    """
    # get the configuration manager
    mgr = tigl3.configuration.CCPACSConfigurationManager_get_instance()

    # get the CPACS configuration, defined by the tigl handle
    # we need to access the underlying tigl handle (that is used in the C/C++ API)
    config = mgr.get_configuration(tigl_handle._handle.value)
    uid_manager = config.get_uidmanager()

    f1 = uid_manager.get_geometric_component("Fuselage")

    structure = f1.get_structure()
    skin = structure.get_skin()  # Funktioniert nicht

    print(dir(skin))

if __name__ == '__main__':
    tixi_h = tixi3wrapper.Tixi3()
    tigl_h = tigl3wrapper.Tigl3()

    dir_path = os.path.dirname(os.path.realpath(__file__))
    tixi_h.open(dir_path + "/fuselage_structure-v3.xml")
    tigl_h.open(tixi_h, "")

    display_fuselage_structure(tigl_h)
```

joergbrech, 2024-11-27:
The class is probably just not properly registered in the configuration files for the Python bindings with swig. Thanks for letting us know! We will check it out.

hardt-fa-flb, 2024-12-10:
List of broken python bindings:

* `CCPACSFuselageStructure.GetSkin()`, and all other `boost::optional` bindings
* `CCPACSSkin.GetSkinSegments()`
* `CPACSSkinSegments.GetSkinSegments()`
* `CCPACSSkinSegment.GetGeometry()`, via `CTiglStringerFrameBorderedObject`, which is ignored in the SWIG config

daha96, 2025-01-13:
The `boost::optional` bindings bug includes also **all fuselage structure components** like frames, stringers, doors, walls, ....

closed 2026-08-31 by joergbrech