DLR-SC/tigl #1069: Failed to start tiglviewer-3 due to MESA_GLSL_CACHE_DISABLE

BeccyB, 2025-03-11:
After building on linux with

cmake ..              -DCMAKE_BUILD_TYPE=Release              -DCMAKE_INSTALL_PREFIX=install   -DOCE_STATIC_LIBS=ON              -DTIGL_BUILD_TESTS=ON -DCMAKE_CXX_FLAGS=-fno-stack-protector

I get the following error:

*** MESA_GLSL_CACHE_DISABLE is deprecated; use MESA_SHADER_CACHE_DISABLE instead ***
*** stack smashing detected ***: terminated
Aborted

After adding the flag -DTIGL_BUILD_TESTS=ON -DCMAKE_CXX_FLAGS=-fno-stack-protector I now see a short 1 sec glimps of a gui before crashing.

```
# packages in environment at /home/bryd_re/.conda/envs/tigl-bld:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
brotlipy                  0.7.0           py38h01eb140_1006    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
ca-certificates           2025.1.31            hbcca054_0    conda-forge
cairo                     1.18.4               h3394656_0    conda-forge
cffi                      1.17.0           py38heb5c249_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
doxygen                   1.8.15             hf484d3e_100    dlr-sc
expat                     2.6.4                h5888daf_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h7e30c49_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimageplus             3.18.0               hf484d3e_2    dlr-sc
freetype                  2.12.1               h267a509_2    conda-forge
gettext                   0.23.1               h5888daf_0    conda-forge
gettext-tools             0.23.1               h5888daf_0    conda-forge
giflib                    5.2.2                hd590300_0    conda-forge
glib                      2.82.2               h07242d1_1    conda-forge
glib-tools                2.82.2               h4833e2c_1    conda-forge
graphite2                 1.3.13            h59595ed_1003    conda-forge
gst-plugins-base          1.24.7               h0a52356_0    conda-forge
gstreamer                 1.24.7               hf3bb09a_0    conda-forge
harfbuzz                  9.0.0                hda332d3_1    conda-forge
icu                       75.1                 he02047a_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.21.3               h659f571_0    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
ld_impl_linux-64          2.43                 h712a8e2_4    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libasprintf               0.23.1               h8e693c7_0    conda-forge
libasprintf-devel         0.23.1               h8e693c7_0    conda-forge
libcap                    2.75                 h39aace5_0    conda-forge
libclang-cpp15            15.0.7          default_h127d8a8_5    conda-forge
libclang13                19.1.7          default_h9c6a7e4_1    conda-forge
libcups                   2.3.3                h4637d8d_4    conda-forge
libdeflate                1.23                 h4ddbbb0_0    conda-forge
libdrm                    2.4.124              hb9d3cd8_0    conda-forge
libedit                   3.1.20250104    pl5321h7949ede_0    conda-forge
libegl                    1.7.0                ha4b6fd6_2    conda-forge
libevent                  2.1.12               hf998b51_1    conda-forge
libexpat                  2.6.4                h5888daf_0    conda-forge
libffi                    3.4.6                h2dba641_0    conda-forge
libflac                   1.4.3                h59595ed_0    conda-forge
libgcc                    14.2.0               h767d61c_2    conda-forge
libgcc-ng                 14.2.0               h69a702a_2    conda-forge
libgcrypt-lib             1.11.0               hb9d3cd8_2    conda-forge
libgettextpo              0.23.1               h5888daf_0    conda-forge
[... 68 lines cut ...]
xcb-util-image            0.4.0                hb711507_2    conda-forge
xcb-util-keysyms          0.4.1                hb711507_0    conda-forge
xcb-util-renderutil       0.3.10               hb711507_0    conda-forge
xcb-util-wm               0.4.2                hb711507_0    conda-forge
xkeyboard-config          2.43                 hb9d3cd8_0    conda-forge
xorg-libice               1.1.2                hb9d3cd8_0    conda-forge
xorg-libsm                1.2.5                he73a12e_0    conda-forge
xorg-libx11               1.8.11               h4f16b4b_0    conda-forge
xorg-libxau               1.0.12               hb9d3cd8_0    conda-forge
xorg-libxcomposite        0.4.6                hb9d3cd8_2    conda-forge
xorg-libxdamage           1.1.6                hb9d3cd8_0    conda-forge
xorg-libxdmcp             1.1.5                hb9d3cd8_0    conda-forge
xorg-libxext              1.3.6                hb9d3cd8_0    conda-forge
xorg-libxfixes            6.0.1                hb9d3cd8_0    conda-forge
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
xorg-libxxf86vm           1.1.6                hb9d3cd8_0    conda-forge
xorg-xf86vidmodeproto     2.3.1             hb9d3cd8_1005    conda-forge
xz                        5.6.4                hbcc6ac9_0    conda-forge
xz-gpl-tools              5.6.4                hbcc6ac9_0    conda-forge
xz-tools                  5.6.4                hb9d3cd8_0    conda-forge
zstd                      1.5.7                hb8e6e7a_1    conda-forge
```

BeccyB, 2025-03-11:
[tigl_conda.txt](https://github.com/user-attachments/files/19181463/tigl_conda.txt)

BeccyB, 2025-03-11:
The problem seems to be the python env. It works with this older version: 

[tigl_conda.txt](https://github.com/user-attachments/files/19181708/tigl_conda.txt)

svengoldberg, 2025-03-24:
This problem results from naming conflicts. The newer conda environment installs its own GL (because of a newer QT). The used system already has GL installed. Therefore, we use the old QT version. That should fix this issue.