DLR-SC/tigl #1073: Continuous fuselage lofting may cause overshoots

sdeinert, 2025-03-28:
The attached CPACS model causes an overshoot of the last segment's loft surface:

[image]

Is there a way to prevent this?

Since xml is not allowed as an attachment, here's the file content:
```
<?xml version="1.0" encoding="utf-8"?>
<cpacs xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
  <header>
    <name>Descartes Model</name>
    <version>1.0</version>
    <cpacsVersion>3.3</cpacsVersion>
    <creator>Descartes</creator>
    <description>CPACS aircraft model designed using Descartes</description>
    <timestamp>2019-11-14T12:11:18</timestamp>
  </header>
  <vehicles>
    <aircraft>
      <model uID="model">
        <name>model</name>
        <fuselages>
          <fuselage uID="Fuselage" loftContinuity="C02">
            <name>Fuselage</name>
            <transformation uID="Fuselage_transformation1">
              <scaling uID="Fuselage_transformation1_scaling1">
                <x>1</x>
                <y>1</y>
                <z>1</z>
              </scaling>
              <rotation uID="Fuselage_transformation1_rotation1">
                <x>0</x>
                <y>0</y>
                <z>0</z>
              </rotation>
              <translation uID="Fuselage_transformation1_translation1">
                <x>0</x>
                <y>0</y>
                <z>0</z>
              </translation>
            </transformation>
            <sections>
              <section uID="Fuselage_section8">
                <name>Fuselage_section8</name>
                <transformation uID="Fuselage_section8_transformation1">
                  <scaling uID="Fuselage_section8_transformation1_scaling1">
                    <x>1</x>
                    <y>1.84</y>
                    <z>1.98</z>
                  </scaling>
                  <rotation uID="Fuselage_section8_transformation1_rotation1">
                    <x>0</x>
                    <y>0</y>
                    <z>0</z>
                  </rotation>
                  <translation uID="Fuselage_section8_transformation1_translation1" refType="absLocal">
                    <x>3.15789</x>
                    <y>0</y>
                    <z>0.14</z>
                  </translation>
                </transformation>
                <elements>
                  <element uID="Fuselage_section8_element1">
                    <name>Fuselage_section8_element1</name>
                    <profileUID>fuse_superellipse1</profileUID>
                    <transformation uID="Fuselage_section8_element1_transformation1"/>
                  </element>
                </elements>
              </section>
[... 227 lines cut ...]
          </fuselage>
        </fuselages>
      </model>
    </aircraft>
    <profiles>
      <fuselageProfiles>
        <fuselageProfile uID="fuse_superellipse1">
          <name>fuse_superellipse1</name>
          <description>A super ellipse profile using two independent super ellipses for upper and lower half of profile with 40 points.</description>
          <pointList>
            <x mapType="vector">0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0;0</x>
            <y mapType="vector">1.97074e-06;0.26941;0.338037;0.384284;0.418835;0.445449;0.465896;0.481131;0.491706;0.49794;0.5;0.495433;0.481755;0.459034;0.427357;0.386792;0.337304;0.278568;0.209498;0.126526;-4.8902e-13;-0.126526;-0.209498;-0.278568;-0.337304;-0.386792;-0.427357;-0.459034;-0.481755;-0.495433;-0.5;-0.49794;-0.491706;-0.481131;-0.465896;-0.445449;-0.418835;-0.384284;-0.338037;-0.26941;1.97074e-06</y>
            <z mapType="vector">-0.5;-0.487764;-0.452254;-0.396946;-0.327254;-0.25;-0.172746;-0.103054;-0.0477458;-0.0122359;0;0.0122359;0.0477458;0.103054;0.172746;0.25;0.327254;0.396946;0.452254;0.487764;0.5;0.487764;0.452254;0.396946;0.327254;0.25;0.172746;0.103054;0.0477458;0.0122359;0;-0.0122359;-0.0477458;-0.103054;-0.172746;-0.25;-0.327254;-0.396946;-0.452254;-0.487764;-0.5</z>
            <kinkIndices mapType="vector">30;10</kinkIndices>
            <parameterMap>
              <pointIndices mapType="vector">10;20;30</pointIndices>
              <paramOnCurve mapType="vector">0.25;0.5;0.75</paramOnCurve>
            </parameterMap>
          </pointList>
        </fuselageProfile>
      </fuselageProfiles>
    </profiles>
  </vehicles>
</cpacs>
```

rainman110, 2025-03-28:
The cause is a very uneven section distribution. Removing section 14 fixes it.

rainman110, 2025-03-28:
Alternatively insert one more section into the long segment (segment 13 I believe)

joergbrech, 2025-03-28:
> Alternatively insert one more section into the long segment (segment 13 I believe)

Yes, that's it. @sdeinert, does this resolve your issue?

[image]

<details>
<summary>Click to toggle CPACS with additional section, created with CPACS Creator 0.2</summary>

```xml
<?xml version="1.0" encoding="utf-8"?>
<cpacs xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
  <header>
    <name>Descartes Model</name>
    <version>1.0</version>
    <cpacsVersion>3.3</cpacsVersion>
    <creator>Descartes</creator>
    <description>CPACS aircraft model designed using Descartes</description>
    <timestamp>2019-11-14T12:11:18</timestamp>
    <versionInfos/>
  </header>
  <vehicles>
    <aircraft>
      <model uID="model">
        <name>model</name>
        <fuselages>
          <fuselage uID="Fuselage" loftContinuity="C02">
            <name>Fuselage</name>
            <transformation uID="Fuselage_transformation1">
              <scaling uID="Fuselage_transformation1_scaling1">
                <x>1</x>
                <y>1</y>
                <z>1</z>
              </scaling>
              <rotation uID="Fuselage_transformation1_rotation1">
                <x>0</x>
                <y>0</y>
                <z>0</z>
              </rotation>
              <translation uID="Fuselage_transformation1_translation1">
                <x>0</x>
                <y>0</y>
                <z>0</z>
              </translation>
            </transformation>
            <sections>
              <section uID="Fuselage_section8">
                <name>Fuselage_section8</name>
                <transformation uID="Fuselage_section8_transformation1">
                  <scaling uID="Fuselage_section8_transformation1_scaling1">
                    <x>1</x>
                    <y>1.84</y>
                    <z>1.98</z>
                  </scaling>
                  <rotation uID="Fuselage_section8_transformation1_rotation1">
                    <x>0</x>
                    <y>0</y>
                    <z>0</z>
                  </rotation>
                  <translation uID="Fuselage_section8_transformation1_translation1" refType="absLocal">
                    <x>3.15789</x>
                    <y>0</y>
                    <z>0.14</z>
                  </translation>
                </transformation>
                <elements>
                  <element uID="Fuselage_section8_element1">
                    <name>Fuselage_section8_element1</name>
                    <profileUID>fuse_superellipse1</profileUID>
                    <transformation uID="Fuselage_section8_element1_transformation1"/>
[... 303 lines cut ...]
              <paramOnCurve mapType="vector">0.25;0.5;0.75</paramOnCurve>
            </parameterMap>
          </pointList>
        </fuselageProfile>
        <fuselageProfile uID="fuse_superellipse2">
          <name>Superellipse3</name>
          <description>Standard Profile Type Superellipse</description>
          <standardProfile>
            <superEllipse>
              <lowerHeightFraction>0.5</lowerHeightFraction>
              <mLower>2</mLower>
              <mUpper>3</mUpper>
              <nLower>1</nLower>
              <nUpper>3</nUpper>
            </superEllipse>
          </standardProfile>
        </fuselageProfile>
      </fuselageProfiles>
    </profiles>
  </vehicles>
</cpacs>

```

</details>

sdeinert, 2025-03-28:
Ok thanks for the hints!
I also noticed that it depends on the distribution and number of sections, but didn't see this as a solution more as a workaround.
From your answers I gather this cannot be solved on the code side?

rainman110, 2025-03-28:
This is how bspline interpolation works. If the points/sections are very uneven, the result can oscillate. From the code side, one can play around with the interpolation parameters, but then you 'll get problems elsewhere. So the best way is to ensure an even distribution of the sections.

closed 2025-04-24 by joergbrech