DLR-SC/tigl #1144: CPACS Creator Bug when changing Profile for one section

ole-alb, 2025-08-27:
It is not possible to change the Profile "Profile UiD" of a wing section or fuselage section. Only possible for entire wing or fuselage. 

Error Message: Number of edges not equal in CTiglMakeLoft

joergbrech, 2025-08-28:
If you mix different profiles, all profiles must have either a sharp or a blunt trailing edge, you are not allowed to mix sharp and blunt profiles.

The only profile in the DB with a blung trailing edge is NACA0012. So changing any single section to a profile with a sharp trailing edge gives the error. 

It is possible to first change the profile for the entire wing to a profile with a sharp trailing edge, and then choose different profiles for the different sections *(As long as it is not NACA0012)*.

Therefore I would say this issue is fixed by a better error message.

joergbrech, 2025-08-28:
I noticed another problem: When the change of a profile fails and I undo the change and subsequently choose another profile, CPACS Creator crashes. The reason for this is that the profilesDB object references the old cpacs configuration, which gets overwritten by the Undo Command. Therefore when we update the configuration in the ModificatorManager, we have to make sure that the profilesDB points to the profiles of the new configuration.

closed 2025-08-28 by joergbrech